YosysHQ/apicula #300: Proposal to re-enable ALU mode

aerkiaga, 2024-12-21:
If I correctly understand, #78 disables ALU mode 8 (`CUPCDN`), since it has the same `LUT4` bits as mode 1 (`SUB`) and therefore can't be correctly identified by `gowin_unpack`. An alternative option was mentioned, namely defaulting to `SUB`. I would like to propose setting the 4 unused bits (8 through 11) in `CUPCDN` to some value in `gowin_pack` so that `gowin_unpack` can then recover this mode if produced with that tool, and only defaulting to `SUB` if the usual value is found.

I would like your opinion on the matter before sending any changes. I would also like your opinion on adding an extra mode implementing the functionality described [at the end of this page](https://github.com/YosysHQ/apicula/blob/master/doc/alu.md), which could complement mode `9` for implementing a multiplier.

aerkiaga, 2024-12-21:
If I understand correctly, this is how `nextpnr` wires the ALU:
```
add(0)    0011000011001100  A:-  B:I1 C:1 D:I0
sub(1)    1010000001011010  A:I0 B:-  C:1 D:I1
addsub(2) 0110000010011010  A:I0 B:I1 C:1 D:I3
ne(3)     1001000010011111  A:I0 B:I1 C:1 D:-
ge(4)     1001000010011010  A:I0 B:I1 C:1 D:-
le(5)     N/A
cup(6)    1010000010100000  A:I0 B:I1 C:1 D:-
cdn(7)    0101000001011111  A:I0 B:I1 C:1 D:-
cupcdn(8) N/A
mul(9)    0111100010001000  A:I0 B:I1 C:1 D:I3
```

So, that answers my second question; the `MUL` mode is already doing what the page describes.

pepijndevos, 2024-12-21:
As mentioned in the other issue I don't have particularly strong feelings about the modes yosys isn't using. As long as CI is green and users are happy I'm happy. So I'd happily take a pr that reenables this mode.

Iirc there is some unde the hood reshuffling of ports so as a user it's not as mixed up as in the primitives

aerkiaga, 2024-12-21:
@pepijndevos I see `AND` ports are swapped [here](https://github.com/YosysHQ/nextpnr/blob/f01465f6280e2eae2459819597fda0fc2f46b703/himbaechel/uarch/gowin/pack.cc#L1420). You mean more swapping is done elsewhere, for `SUB` and `MUL`? Is that done in `yosys`, `nextpnr` itself or `gowin_pack`?

pepijndevos, 2024-12-22:
I think that might be it

PR #301 (2024-12-22): Restore CUPCDN with a different bit pattern
This should suffice to get `CUPCDN` working. Closes #300.
closed 2024-12-22 by pepijndevos