YosysHQ/apicula #302: noob here - No BELs remaining.

raymonds2019, 2024-12-29:
This might not be a bug. 
Maybe there is a irc channel for this question, please let me know.

Working on the dvi himbaechel example for 9k port to 4k.
ERROR: Unable to place cell 'clk_div', no BELs remaining to implement cell type 'CLKDIV'

I am thinking this has to do with 4k has two quadrants and 9k has 4, could this be explained?

yrabbit, 2024-12-29:
Well the channels for communication (and IRC including) are described in the main README and on the first page of https://github.com/YosysHQ/apicula#resources.

And currently CLKDIV is supported only on Tangnano9k, Tangnano20k, TangPrimer20k boards.

raymonds2019, 2024-12-29:
Ok thank you. Please close this.

closed 2024-12-29 by yrabbit