YosysHQ/apicula #309: Fuzzing does not work with Gowin Software 1.9.9 and 1.9.11

JR-Elo, 2025-02-15:
Description
1. The fuzzing process fails
2. Chip DB (pickle files) can only be created with Gowin IDE 1.9.8. Using newer versions (1.9.9, 1.9.11) fails

Steps to reproduce
1. Clone apicula repository
2. Set up environment with Gowin IDE (tested with 1.9.8, 1.9.9, 1.9.11)
3. Run fuzzer

`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/fpga/git/apicula
  Preparing metadata (setup.py) ... done
Requirement already satisfied: crc in /home/pi/Programme/apycula_env/lib/python3.11/site-packages (from Apycula==0.16.dev37+gce70be7) (7.0.0)
Installing collected packages: Apycula
  Attempting uninstall: Apycula
    Found existing installation: Apycula 0.12
    Uninstalling Apycula-0.12:
      Successfully uninstalled Apycula-0.12
  Running setup.py develop for Apycula
Successfully installed Apycula-0.16.dev37+gce70be7
Fuzzing wird gestartet...
python3 -m apycula.tiled_fuzzer GW1N-1
check 19199191
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in runmodule_as_main
  File "<frozen runpy>", line 88, in runcode
  File "/home/pi/fpga/git/apicula/apycula/tiled_fuzzer.py", line 237, in <module>
    tm = tm_h4x.read_tm(f, device)
         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/fpga/git/apicula/apycula/tm_h4x.py", line 318, in read_tm
    assert len(chunk) == chunklen
           ^^^^^^^^^^^^^^^^^^^^^^
AssertionError
make: *** [Makefile:12: GW1N-1_stage1.pickle] Error 1`

The Tang Nano 9K is becoming increasingly popular in the FPGA community, and proper support for current IDE versions would be very helpful.

Chandler-Kluser, 2025-02-15:
> Description:
> 
>     1. The fuzzing process fails with an Error when using Gowin IDE versions 1.9.9 and 1.9.11
> 
>     2. Chip database (pickle files) generation only works with Gowin IDE 1.9.8, but fails with newer versions
> 
> 
> Steps to reproduce:
> 
>     1. Clone apicula repository
> 
>     2. Set up environment with Gowin IDE (tested with 1.9.8, 1.9.9, 1.9.11)
> 
>     3. Run fuzzer
> 
> 
> `Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple Obtaining file:///home/pi/fpga/git/apicula Preparing metadata (setup.py) ... done Requirement already satisfied: crc in /home/pi/Programme/apycula_env/lib/python3.11/site-packages (from Apycula==0.16.dev37+gce70be7) (7.0.0) Installing collected packages: Apycula Attempting uninstall: Apycula Found existing installation: Apycula 0.12 Uninstalling Apycula-0.12: Successfully uninstalled Apycula-0.12 Running setup.py develop for Apycula Successfully installed Apycula-0.16.dev37+gce70be7 Fuzzing wird gestartet... python3 -m apycula.tiled_fuzzer GW1N-1 check 19199191 Traceback (most recent call last): File "<frozen runpy>", line 198, in runmodule_as_main File "<frozen runpy>", line 88, in runcode File "/home/pi/fpga/git/apicula/apycula/tiled_fuzzer.py", line 237, in <module> tm = tm_h4x.read_tm(f, device) ^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/pi/fpga/git/apicula/apycula/tm_h4x.py", line 318, in read_tm assert len(chunk) == chunklen ^^^^^^^^^^^^^^^^^^^^^^ AssertionError make: *** [Makefile:12: GW1N-1_stage1.pickle] Error 1`
> 
> The Tang Nano 9K is becoming increasingly popular in the FPGA community, and proper support for current IDE versions would be very helpful.
> 
> Environment:
> 
>     * OS: Pop!_OS 22.04 LTS
> 
>     * Kernel: 6.9.3-76060903-generic
> 
>     * Architecture: x86_64
> 
>     * Python: 3.10.12
> 
>     * Apicula: 0.16.dev37+gce70be7
> 
>     * Gowin IDE versions tested: 1.9.8, 1.9.9, 1.9.11
> 
> 
> NOTE: Version v1.9.8 (http://cdn.gowinsemi.com.cn/Gowin_V1.9.8_linux.tar.gz) is no longer available for download.

I have `Gowin_V1.9.8_Education_linux.tar.gz`, `Gowin_V1.9.8.03_Education.tar.gz` and `Gowin_V1.9.8.09_Education_linux.tar.gz` if you want

yrabbit, 2025-02-16:
Just to clarify what the snag is with the quick transition to the new version. Gowin IDE stores its data by writing directly from memory structures - imagine 'struct bla_bla_bla -> take address &bla_bla_bla and write to file fwrite(&bla_bla_bla_bla, sizeof(bla-bla_bla), 1, file)'.
If you add a byte-sized field in the middle of the structure, even an unused one, you can write this data and read it without problems, but for third-party programs that are not aware of this additional byte, everything will break catastrophically - the structure fields are no longer in place and garbage will be read from the file.
Further Gowin can add and remove fields even every week and we will know about it only by crash because nobody shares sources.
Support for the new version is to meditate on the binary file and hypothesize what they did this time and where to look for the meaning.
This work is only justified if the old IDE simply does not support the chip family, which is not true for Tangnano9k. (well there are a couple other considerations).
Of course it would be great if you added support for 1.9.11 :)
(only quickly because with the release of 1.9.12 all work will be thrown out the window).

Chandler-Kluser, 2025-02-16:
if Tang Nano 20k, 9k, 4k as well as tang primer 20k are working fine (as far as I tested, they are), the next big challenge is to have a better placement according to time constraints because Gowin pnr tool is a mess for that

Chandler-Kluser, 2025-02-16:
@yrabbit btw thank you for the awesome work done

pepijndevos, 2025-02-16:
Fwiw I think @awooOOoo upgraded the ide support in #273 but it seems work on gw5a stalled a bit