YosysHQ/apicula #312: cannot find boardname.pickle when doing nextpnr-himbaechel install

oltessis, 2025-02-18:
I am trying to install nextpnr-himbaechel, I had some module not found errors so I moved the apycula folder to nextpnr/himbaechel/uarch/gowin. Now the installer cannot find .pickle files where it is looking for GW1NS-4.pickle, etc...

yrabbit, 2025-02-18:
I'd like to see the original error message, and what OS you have.

Alternatively uninstall everything, do `pip install apicula` and try installing nextpnr again

oltessis, 2025-02-19:
[image]

I am using debian 12

[image]

yrabbit, 2025-02-19:
I was wrong - I downloaded Debian 12, installed it on a clean hard drive, installed the right stuff for nextpnr and the sources themselves. 

``` shell
rabbit@grf:~$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Found that Debian says bad words on `pip install apycula` :  something about "error: externally-managed-environment"

I've never seen anything like this in my life under BSD. I compiled nextpnr anyway, but only because I don't like problems being created out of nowhere.

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
HIGHLY NOT RECOMMENDED, it's better to wait for someone who knows Python and Debian to answer.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

But if you really want to:

``` shell
pipx install apycula
mkdir src
cd src
git clone https://github.com/YosysHQ/nextpnr.git
cd nextpnr
git submodule update --init --recursive
mkdir -p build
cd build
cmake .. -DARCH="himbaechel" -DHIMBAECHEL_UARCH="gowin"
env PYTHONPATH=${HOME}/.local/pipx/venvs/apycula/lib/python3.11/site-packages/ make
```

pepijndevos, 2025-02-19:
Externally managed means it's trying to install to where Debian installs its python packages.

```
python -m venv env
source env/bin/activate
pip install apycula
```

uis246, 2025-02-21:
> Found that Debian says bad words on pip install apycula : something about "error: externally-managed-environment"

https://wiki.gentoo.org/wiki/Pip#Package_installation
https://peps.python.org/pep-0668/
Means pip sees system package manager.