YosysHQ/apicula #322: ` gowin_pll`  does not find working configuration while GOWIN IDE can?

GyrosGeier, 2025-03-29:
On this [question](https://electronics.stackexchange.com/q/742342) on StackOverflow there is a comment that the GOWIN IDE is able to find a working configuration for an input frequency of 27 MHz and an output frequency of 6 MHz. The ` gowin_pll`  tool does not find this solution, possibly because ranges are different.

yrabbit, 2025-03-29:
I'll look at the acceptable ranges for GW1N-1, although I'm surprised and really want to know where the author managed to buy a board on this chip - let's say on AliExpress is full of tangnano1k, but it's all GW1NZ-1 (the difference in the photo - on the right is Z-1, on the left is the original GW1N-1). :)

[image]

yrabbit, 2025-03-29:
Well, the acceptable range is specified correctly, it's all about the algorithm.

https://github.com/YosysHQ/apicula/blob/ac3bfa0e5d542428808bf5b9e324922b604d9b86/apycula/gowin_pll.py#L51-L59

pepijndevos, 2025-03-29:
What's the actual configuration that Gowin generates? Usually you don't get the exact frequency but the closest fraction within the boundaries.

yrabbit, 2025-03-29:
Well, the IDE worked here for sure: 

𝑓CLKOUT = (𝑓𝐶𝐿𝐾𝐼𝑁 ∗ FBDIV ) ⁄ IDIV = (27 * 2) / 9 = 6 

``` verilog
defparam rpll_inst.FCLKIN = "27";
defparam rpll_inst.DYN_IDIV_SEL = "false";
defparam rpll_inst.IDIV_SEL = 8;
defparam rpll_inst.DYN_FBDIV_SEL = "false";
defparam rpll_inst.FBDIV_SEL = 1;
defparam rpll_inst.DYN_ODIV_SEL = "false";
defparam rpll_inst.ODIV_SEL = 128;
defparam rpll_inst.PSDA_SEL = "0000";
defparam rpll_inst.DYN_DA_EN = "true";
defparam rpll_inst.DUTYDA_SEL = "1000";
defparam rpll_inst.CLKOUT_FT_DIR = 1'b1;
defparam rpll_inst.CLKOUTP_FT_DIR = 1'b1;
defparam rpll_inst.CLKOUT_DLY_STEP = 0;
defparam rpll_inst.CLKOUTP_DLY_STEP = 0;
defparam rpll_inst.CLKFB_SEL = "internal";
defparam rpll_inst.CLKOUT_BYPASS = "false";
defparam rpll_inst.CLKOUTP_BYPASS = "false";
defparam rpll_inst.CLKOUTD_BYPASS = "false";
defparam rpll_inst.DYN_SDIV_SEL = 2;
defparam rpll_inst.CLKOUTD_SRC = "CLKOUT";
defparam rpll_inst.CLKOUTD3_SRC = "CLKOUT";
defparam rpll_inst.DEVICE = "GW1N-1";
```

vstrakh, 2025-03-29:
The Tang Nano boards has C6/I5 speed class chips (both GW1N-1 and GW1NZ-1), but @GyrosGeier used C5/I4 class with apicula tools. Maybe it's just misunderstanding with command line options given?

GyrosGeier, 2025-03-29:
Yes, also I get the same result for C6/I5

closed 2025-03-31 by yrabbit