YosysHQ/apicula #337: gowin_bba should be gowin_bbva in setup.py?

Apteryks, 2025-05-20:
Hello,

The Guix sanity checker found that the line below does not match an actual module:

https://github.com/YosysHQ/apicula/blob/989c6c106fd411f348b6770c368771fe08b4a23c/setup.py#L31

The error looks like:
```
ModuleNotFoundError: No module named 'apycula.gowin_bba'
```

Renaming it to gowin_bbva resolved the issue.

yrabbit, 2025-05-21:
I'm not an expert here, but bba is no longer needed for nextpnr-himbaechel, so how about removing all mention of it? Would that help?

Apteryks, 2025-05-21:
That would solve the issue, yes.  Going farther in the cleanup, perhaps the gowin_bbva module could be removed as well, if it was related.

yrabbit, 2025-05-21:
Will you do a PR? Because I don't even know what bbva is.

[image]

Apteryks, 2025-05-21:
Ah, sorry, your are right, it doesn't exist. It can simply be removed from the setup.py then :-).

PR #338 (2025-05-21): Remove a non-existent module from setup.
Fixes #337
commit 9994858a (2025-05-21): Remove a non-existent module from setup.

Fixes #337
closed 2025-05-21 by yrabbit