YosysHQ/apicula #353: Question about DSP, way to use it with pre-adders and cascading as single big block, and cascading those ?

Pe3ucTop, 2025-06-27:
Hello all.

Came here because looking for alternative for GOWIN FPGA Designer, because it lack in DSP full potential use.

Short about use case:
- implementing 1d signal filter - Gaussian blur (it's center symmetrical)
- realize convolution with cascaded DSPs 
- optimize/minimize multiplication count by using pre-added (signal input values left/right window side)
Project worked till here :)

Came to idea to use DSP(pre-adder) cascade and shift  register to minimize total resource use and here start problems.

Both propriety soft and here I see split use of pre-adder PADD18 and multiplier/alu MULTADDALU18X18 , and problem comes from shared cascade resource ?
- Both blocks (PADD18/MULTADDALU18X18 ) report to have "A" cascading (SI/SO and SIA/SOA) , but from unit documentation it's one for both.
- PADD18 primitive have parameter to enable cascade output register (REGSD), like each would have own, but it's only one per whole DSP block and also get used from MULTADDALU18X18 .

[image]

Questions: 
Any ideas to workaround this problem or use with existing primitives ? 
Is there way to use whole DSP block with Apicula ?
Is there way to connect some "parameters" to project "variable/register" like pre-adder Add/Sub selection be able after synthesis ? 

I can provide test project if it would help :)

yrabbit, 2025-06-27:
I won't say for synthesis - not my field at all, but the bad news is that you can't push PADD and multiplier into one DSP block in Apicula at least because PADD is implemented as addition followed by multiplication by one, i.e. the multiplier is occupied.

Gowin IDE can combine separate PADD and MULT primitives into one block - I saw this a year ago (?). But you have to be careful about connections - let's say the result of addition can't come out of the DSP block in pure form in any way - PADD is always connected to the input MUXes of multipliers and only to them.

Pe3ucTop, 2025-06-27:
Oh, it's sad to hear, then I do not have a choice and should wait for response from Gowin support.

So, you are saying that PADD is even abused (hard connected) with MULT ? 
If I understand you correct it's not synthesis, but PnR (nextpnr himbaechel) or there is more steps in the middle ?