YosysHQ/apicula #355: ERROR: Unconstrained IO:o_OBUF_O_99

brandonros, 2025-07-08:
```
(.venv) root@39bb68e5-611f-40c9-91aa-138dbfc8edee:/mnt/build/apicula/examples# ../../nextpnr-himbaechel --json ../../top.json \
                   --write ../../pnrtop.json \
                   --device $DEVICE \
                   --vopt family=GW2A-18C \
                   --vopt cst=$BOARD.cst
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'

Info: Cell O_sdram_clk not found
Info: Cell O_sdram_cke not found
Info: Cell O_sdram_cs_n not found
Info: Cell O_sdram_cas_n not found
Info: Cell O_sdram_ras_n not found
Info: Cell O_sdram_wen_n not found
Info: Cell O_sdram_dqm[0] not found
Info: Cell O_sdram_dqm[1] not found
Info: Cell O_sdram_dqm[2] not found
Info: Cell O_sdram_dqm[3] not found
Info: Cell O_sdram_addr[0] not found
Info: Cell O_sdram_addr[1] not found
Info: Cell O_sdram_addr[2] not found
Info: Cell O_sdram_addr[3] not found
Info: Cell O_sdram_addr[4] not found
Info: Cell O_sdram_addr[5] not found
Info: Cell O_sdram_addr[6] not found
Info: Cell O_sdram_addr[7] not found
Info: Cell O_sdram_addr[8] not found
Info: Cell O_sdram_addr[9] not found
Info: Cell O_sdram_addr[10] not found
Info: Cell O_sdram_ba[0] not found
Info: Cell O_sdram_ba[1] not found
Info: Cell IO_sdram_dq[0] not found
Info: Cell IO_sdram_dq[1] not found
Info: Cell IO_sdram_dq[2] not found
Info: Cell IO_sdram_dq[3] not found
Info: Cell IO_sdram_dq[4] not found
Info: Cell IO_sdram_dq[5] not found
Info: Cell IO_sdram_dq[6] not found
Info: Cell IO_sdram_dq[7] not found
Info: Cell IO_sdram_dq[8] not found
Info: Cell IO_sdram_dq[9] not found
Info: Cell IO_sdram_dq[10] not found
Info: Cell IO_sdram_dq[11] not found
Info: Cell IO_sdram_dq[12] not found
Info: Cell IO_sdram_dq[13] not found
Info: Cell IO_sdram_dq[14] not found
Info: Cell IO_sdram_dq[15] not found
Info: Cell IO_sdram_dq[16] not found
Info: Cell IO_sdram_dq[17] not found
Info: Cell IO_sdram_dq[18] not found
Info: Cell IO_sdram_dq[19] not found
Info: Cell IO_sdram_dq[20] not found
Info: Cell IO_sdram_dq[21] not found
Info: Cell IO_sdram_dq[22] not found
Info: Cell IO_sdram_dq[23] not found
Info: Cell IO_sdram_dq[24] not found
Info: Cell IO_sdram_dq[25] not found
Info: Cell IO_sdram_dq[26] not found
Info: Cell IO_sdram_dq[27] not found
Info: Cell IO_sdram_dq[28] not found
Info: Cell IO_sdram_dq[29] not found
Info: Cell IO_sdram_dq[30] not found
Info: Cell IO_sdram_dq[31] not found
Info: Reading constraints...
Info: Cell clk not found
Info: Cell clk not found
Info: Net clk not found
Info: Cell led[0] not found
Info: Cell led[0] not found
Info: Cell led[1] not found
Info: Cell led[1] not found
Info: Cell led[2] not found
Info: Cell led[2] not found
Info: Cell led[3] not found
Info: Cell led[3] not found
Info: Cell led[4] not found
Info: Cell led[4] not found
Info: Cell led[5] not found
Info: Cell led[5] not found
Info: Cell clk_i not found
Info: Cell clk_i not found
Info: Net clk_i not found
Info: Cell key_i not found
Info: Cell key_i not found
Info: Cell rst_i not found
Info: Cell rst_i not found
Info: Cell TXD not found
Info: Cell TXD not found
Info: Cell RXD not found
Info: Cell RXD not found
Info: Cell LED_R not found
Info: Cell LED_R not found
Info: Cell LED_G not found
Info: Cell LED_G not found
Info: Cell LED_B not found
Info: Cell LED_B not found
Info: Cell tlvds_p not found
Info: Cell tlvds_p not found
Info: Cell tlvds_n not found
Info: Cell tlvds_n not found
Info: Cell elvds_p not found
Info: Cell elvds_p not found
Info: Cell elvds_n not found
Info: Cell elvds_n not found
Info: Cell led[6] not found
Info: Cell led[7] not found
Info: Cell oser_out not found
Info: Cell oser_out not found
Info: Cell pclk_o not found
Info: Cell pclk_o not found
Info: Cell fclk_o not found
Info: Cell fclk_o not found
Info: Cell fclk_i not found
Info: Cell fclk_i not found
Info: Cell data_i not found
Info: Cell data_i not found
Info: Cell q_o[0] not found
Info: Cell q_o[0] not found
Info: Cell q_o[1] not found
Info: Cell q_o[1] not found
Info: Cell q_o[2] not found
Info: Cell q_o[2] not found
Info: Cell q_o[3] not found
Info: Cell q_o[3] not found
Info: Cell q_o[4] not found
Info: Cell q_o[4] not found
Info: Cell q_o[5] not found
Info: Cell q_o[5] not found
Info: Cell q_o[6] not found
Info: Cell q_o[6] not found
Info: Cell q_o[7] not found
Info: Cell q_o[7] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_CLK not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_DEN not found
Info: Cell LCD_DEN not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell LCD_B[4] not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell div_led not found
Info: Cell div_led not found
Info: Create constant nets...
Info: Modify LUTs...
Info: Pack IOBs...
ERROR: Unconstrained IO:o_OBUF_O_99
0 warnings, 1 error
```

# sha256_fpga
Proof-of-concept SHA256 implementation for FPGA

## How to use

```shell
# generate top.v (rust -> verilog)
cargo run
```

## How to test

```shell
cd test
make && make run
```

## How to flash

```shell
# Sipeed Tang Nano 20K: GW2AR-LV18QN88C8/I7

# install deps
apt-get update && apt-get install -y cmake build-essential libboost-all-dev libeigen3-dev python3-pip python3-venv

# activate python venv
python3 -m venv .venv
source .venv/bin/activate

# pip install
pip install apycula

# cmake
mkdir build
cd build
cmake .. \
  -DARCH="himbaechel" \
  -DHIMBAECHEL_UARCH="gowin"

# yosys
yosys -D LEDS_NR=8 -p "read_verilog top.v; synth_gowin -json top.json"

# nextpnr
DEVICE='GW2AR-LV18QN88C8/I7'
BOARD='tangnano20k'
../../nextpnr-himbaechel --json ../../top.json \
                   --write ../../pnrtop.json \
                   --device $DEVICE \
                   --vopt family=GW2A-18C \
                   --vopt cst=$BOARD.cst

# gowin_pack
gowin_pack -d $DEVICE -o pack.fs pnrtop.json

# flash
openFPGALoader -b $BOARD pack.fs
```

yrabbit, 2025-07-09:
Can we see the verilog of the top level module and the .CST file?

yrabbit, 2025-07-09:
```
module top(input wire [1:0] clock_reset, input wire [512:0] i, output wire [0:0] o);
```
It's a top-level module, right? Then everything described in brackets are physical pins* and should be described in the .CST file.

Like 
```
IO_LOC  "clock_reset[0]" 4;
IO_PORT "clock_reset[0]" IO_TYPE=LVCMOS33 PULL_MODE=NONE;
CLOCK_LOC "clock_reset[0]" BUFG;

IO_LOC  "clock_reset[1]" 88;
IO_PORT "clock_reset[1]" IO_TYPE=LVCMOS33;
``` 
Unfortunately `wire [512:0] i` is a deal breaker as this chip simply does not have that many pins.

*These are not always pins outside the chip, there may be internal wires to SDRAM etc, but they are still assigned a pin name.

brandonros, 2025-07-09:
@yrabbit 100% agree with you, thank you, slowly discovering this

The question now becomes... I need to implement UART basically, right? and all the state required for 8n1 to load in an "input" and output a "result"?

closed 2025-07-09 by brandonros

yrabbit, 2025-07-09:
Well basically yes - you somehow need to feed and pull the results from the FPGA taking into account that there are not enough pins for parallel things ;)

brandonros, 2025-07-09:
Do you think I am setting myself up for pain trying to do Rust -> Verilog with no control over board specific stuff with a seemingly premature library?

yrabbit, 2025-07-09:
Purely personal opinion - an absolute nightmare!

How are you going to debug this thing when something doesn't work? 

I would personally start by making a thing on pure verilog that takes small numbers over UART, adds them up and gives the result.

And knowing my talent I would make a lot of mistakes there, but in the process of debugging I would learn a lot of new things 😆

brandonros, 2025-07-09:
Appreciate the input.

I find it hard to believe there is no "UART Verilog common library" or reusable off the shelf implementation if you will. Any thoughts/criticism to that? I'm not the first person to want to do UART over GPIO pins in Verilog, I'm sure of it lol