YosysHQ/apicula #357: setup/ChipDB error

mark-dawn, 2025-07-13:
Hi,
I was trying to get to this to build on Arch through the aur (through the repo tracking -git package). 
I wouldn't  normally report a failure to the og repo but the setup seems to go smoothly, and the fuzzer seems to be run properly for the G1WN-1, G1WN-9 and G1WN-4
However when it gets to the GW1NS-4 it fails:
```
[snip]
0x4cd500  Table type 69 / 0x45 of size 117
 0 class:C5/I4 len(chunk): 15552 chunklen: 15552
 1 class:C5/I4_LV len(chunk): 15552 chunklen: 15552
 2 class:C6/I5 len(chunk): 15552 chunklen: 15552
 3 class:C6/I5_LV len(chunk): 15552 chunklen: 15552
 4 class:ES len(chunk): 15552 chunklen: 15552
 5 class:ES_LV len(chunk): 15552 chunklen: 15552
 6 class:A4 len(chunk): 15552 chunklen: 15552
 7 class:A4_LV len(chunk): 15552 chunklen: 15552
 8 class:8 len(chunk): 15552 chunklen: 15552
 9 class:9 len(chunk): 15552 chunklen: 15552
10 class:10 len(chunk): 15552 chunklen: 15552
11 class:11 len(chunk): 15552 chunklen: 15552
12 class:C7/I6 len(chunk): 15552 chunklen: 15552
13 class:C7/I6_LV len(chunk): 15552 chunklen: 15552
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/marica/.cache/aurutils/sync/prjapicula-git/src/apicula/apycula/tiled_fuzzer.py", line 296, in <module>
    db = chipdb.from_fse(device, fse, dat)
  File "/home/marica/.cache/aurutils/sync/prjapicula-git/src/apicula/apycula/chipdb.py", line 2621, in from_fse
    fse_create_emcu(dev, device, dat)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "/home/marica/.cache/aurutils/sync/prjapicula-git/src/apicula/apycula/chipdb.py", line 2338, in fse_create_emcu
    make_port(r, c, wire, port, 'EMCU_OUT', outs)
    ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/marica/.cache/aurutils/sync/prjapicula-git/src/apicula/apycula/chipdb.py", line 2299, in make_port
    wire = wirenames[wire]
           ~~~~~~~~~^^^^^^
KeyError: -1
make: *** [Makefile:14: GW1NS-4_stage1.pickle] Error 1
==> ERROR: A failure occurred in build().
    Aborting...
```
I can recreate this also by running the command `GOWINHOME=/[snip]/gowin python3 -m apycula.tiled_fuzzer GW1NS-4` standalone in the repo.
Any idea of what the problem might be?

mark-dawn, 2025-07-15:
Sorry for the delay. I am on the master branch. Python version 3.13.5 if that might make a difference. I can test with another version if you think it might impact it.

yrabbit, 2025-07-15:
I was hoping for an accurate commit hash :)

Let me make it clear - chip bases are created after each merge on GitHub, the results are visible in the commit list:

https://github.com/YosysHQ/apicula/commits/master/

[image]

11/13 is the norm as the two variants will be bad until the nextpnr release.

So I'm looking for more neglected cases and found almost yours - commit 257a4cdf546b0e28c879a2fe199dbf233c089d6b

[image]

So let's see when it was fixed and how - https://github.com/YosysHQ/apicula/commit/d88c9868826bf0e6219290c53f51c21673e828ea

[image]

The reason I'm ranting so much is that you'll probably have to figure out where in your OS to specify using the correct docker image of the correct version:) because this only fixes the generation of GitHub environments, not each specific OS.

closed 2025-07-15 by yrabbit

pepijndevos, 2025-07-16:
Unless aur is dead set on running the fuzzers which requires the gowin ide or docker i'd suggest they pull down https://github.com/YosysHQ/apicula/releases/download/0.0.0.dev/linux-x64-gowin-data.tgz which contains the latest chipdb from git master.