YosysHQ/apicula #358: DVI example not working on TangPrimer20k

UnguidedFreedom, 2025-07-16:
Hi everyone,

First of all thanks for such an awesome effort! 

I have compiled the DVI example in this repository for the TangPrimer20k using `make dvi-example-primer20k.fs` and have flashed it on my board using `openFPGALoader -b tangprimer20k dvi-example-primer20k.fs -f`.

When I do so, my monitor does not display anything, and every few seconds shows a "No signal" message which differs from when either no cable is connected (no message at all), or a cable is connected but with no signal at all (a once off "No signal" message), so it is somewhat picking something up. 

It is worth noting that when I flash the official example `dk_video.fs` (from [here](https://github.com/sipeed/TangPrimer-20K-example/blob/main/HDMI/dk_video.fs)) using `openFPGALoader -b tangprimer20k dk_video.fs -f`, that works fine, so I don't think it is a problem with my board.

I unfortunately do not currently have easy access to the Gowin IDE to try to compile the example with the official tools to compare (to see if the issue is with the Verilog example or with the generated bitstream). 

I have also tried tweaking the code in `pll480.v` to set the `defparam rpll_inst.DEVICE = "GW1N-9C";` to `GW2A-18C` to better match the actual device but it led to the same output.

I would love if someone could help me investigate it, either on their own board or by guiding me through some troubleshooting steps on mine. I do have access to an oscilloscope but the DVI/HDMI pins are not easily exposed to probe as far as I can tell.

UnguidedFreedom, 2025-07-16:
@yrabbit Thank you so much for the quick reply!

I downloaded the artifacts linked and ran the dvi example on my board and it did work with my monitor! I can see that there was a recent release that made changes to TLVDS, I wonder if this might be related, I will try updating the version I'm running to see if I can replicate the working build locally 🤞

closed 2025-07-16 by UnguidedFreedom

cgommel, 2026-09-02:
This is reproducible again on current master (`3328095`, apycula `0.34.dev1+g332809527`, oss-cad-suite 20260902) with a Tang Primer 20K and the Dock ext-board.

`dvi-example-primer20k` builds and loads, but the design does not run at all — `div_led` on C13 stays static. So this is not about a monitor rejecting the signal.

Sipeed's vendor bitstream `HDMI/dk_video.fs` runs on the same board, same cable, same capture device, using the same pins.

Minimal reproducer, unrelated to DVI: a free-running counter driving the LEDs works. Add **one** `TLVDS_OBUF` on `G16/H15` and the counter stops — the LEDs freeze at their reset value. Move the same pair to `K14/K15` and it runs again. The only difference is one line in the constraints.

```verilog
module top (input wire clk, output wire [5:0] led,
            output wire tmds_clk_p, tmds_clk_n);
    reg [23:0] div = 0;
    reg [2:0]  pos = 0;
    always @(posedge clk) begin
        if (div == 24'd6_749_999) begin
            div <= 0;
            pos <= (pos == 3'd5) ? 3'd0 : pos + 3'd1;
        end else div <= div + 1'b1;
    end
    assign led = ~(6'b1 << pos);
    TLVDS_OBUF obuf (.I(div[6]), .O(tmds_clk_p), .OB(tmds_clk_n));
endmodule
```

```
IO_LOC  "clk" H11;
IO_PORT "clk" IO_TYPE=LVCMOS33;
IO_LOC  "tmds_clk_p" G16;          // bank 0 -> frozen; K14 (bank 1) -> works
IO_PORT "tmds_clk_p" IO_TYPE=LVDS25 PULL_MODE=NONE DRIVE=3.5;
IO_LOC  "tmds_clk_n" H15;          // bank 0 -> frozen; K15 (bank 1) -> works
IO_PORT "tmds_clk_n" IO_TYPE=LVDS25 PULL_MODE=NONE DRIVE=3.5;
```

Built with `synth_gowin -family gw2a`, `--vopt family=GW2A-18`, `gowin_pack -c -d GW2A-18`.

`G16/H15` is in bank 0, `K14/K15` in bank 1. The clock input `H11` (IOT27A, GCLKT_0) is in bank 0, and on this board that is unavoidable — the 27 MHz oscillator is wired to H11 and three of the four HDMI pairs of the Dock are in bank 0 as well.

I patched `BankDesc._vcc_ios` so `LVDS25` maps to `3.3` and confirmed the change reaches the bitstream (26 bits differ in the bank configuration region), but the design still does not run — so the bank voltage alone does not explain it.